LordOfMice/hidusbf #198: hidusbf kill my controller help me plz

iTrOzola, 2023-02-03:
When I tried to use hidusbf, my controller is turned off and since Windows no longer detects it,Thank you for helping me

[image]

iTrOzola, 2023-02-03:
[image]

iTrOzola, 2023-02-03:
[image]

LordOfMice, 2023-02-06:
1. Uninstall section of readme.
2. If you posted screenshot why did you select WORKING device rather than non working? What is logic of your action???

closed 2023-02-06 by LordOfMice