LordOfMice/hidusbf #201: Mouse movement causes OS to grind to a halt if overclock is enabled?

mmotti, 2023-03-01:
Hi,

I've overclocked my DualSense controller and noticed that for whatever reason I am getting extreme FPS drops in games if I move my mouse whilst my overclocked controller was still connected. It's as if my entire system locks up.

If I lower the poll rate the issue is less evident but it's still there, however if I untick the "filter on device", disconnect the controller or don't move my mouse I no longer experience the issue.

Please find a video of the issue here: [https://streamable.com/f1cqt0](https://streamable.com/f1cqt0)

I tried to perform a simple task such as opening explorer whilst spamming the mouse and you can see it just locks up until I stop moving it. I also opened explorer without the overclock enabled or when the mouse was stationary and it was fine.

Is there any fix for this?

mmotti, 2023-03-02:
The culprit seems to be GeForce Experience; more specifically the overlay.

I went into safe mode and the issue magically disappeared. Ran DDU as I've had similar issues previously with overlays (be it through Discord or NVIDIA) and the problem was gone.

Re-installed GeForce Experience and the issue returned even before installing the graphics driver.

Disabled the overlay and it no longer happens.

Not sure whether this is an isolated incident or not but might be worth pinning this in case others experience the same issue.

GeForce Experience Version 3.27.0.112

LordOfMice, 2023-03-02:
>this is an isolated incident 

This is issue, but not with THIS software. The problem is not new and usually described by "after overclocking when I move my mouse I see higher CPU consuming". But most people don't understand well what they are actually measuring. And that is not overclocking of device (CPU consuming of overclocking itself is so tiny, that simply can not be measured at all by such tools as Task Manager) rather than reaction of some SOFTWARE to this overclocking (and if you move mouse over desktop this software is usually Explorer).

>but might be worth pinning this

You can not pin more than three issues and аnd now the hottest ones are pinned of course. But i will refer this on wiki page as typical example.

closed 2023-03-02 by LordOfMice

alexanderadreon, 2023-05-28:
> Hi,
>
> I've overclocked my DualSense controller and noticed that for whatever reason I am getting extreme FPS drops in games if I move my mouse whilst my overclocked controller was still connected. [...]

I read through the thread and im not understanding how to fix the problem can i get some help?

Drift91, 2023-08-25:
@mmotti That actually explains so much. I'd noticed that the hitching would randomly come and go. I'd try a bunch of things to fix it, test, and it was fine. I'd start reinstalling apps one by one, testing after each, and it was fine still. Then later on I'd go to actually play a game and I'd find it'd returned again. What I didn't actually think about was that I'd randomly find the Geforce Experience overlay disabled and re-enable it, and I suspect the game that's been disabling it is one of the first games I'd use to test the stuttering most of the time. I'll have to try keeping the overlay disabled and see if that helps. If not I'm going to rip GFE right out since I hate the bloated crapware app to begin with and I only have it installed so I can use Freestyle effects in a single game that's incompatible with Reshade.

@alexanderadreon Try turning off the Geforce Experience overlay, if you don't have the overlay enabled or Geforce Experience installed then some other app or driver is causing your frame time hitches. Another app that was causing stuttering/hitches for me was RivaTuner Statistics Server; In my testing RTSS only caused stuttering when it's overlay keybind was enabled through MSI Afterburner, but others have suggested completely uninstalling it is necessary to alleviate the issue.

By the way, as LordOfMice was saying, having your mouse on a higher USB polling rate, whether via overclocking or natively on a more expensive gaming mouse, will generally increase CPU usage due to there being more mouse updates that need to be processed. [This article](https://blogs.windows.com/windowsdeveloper/2023/05/26/delivering-delightful-performance-for-more-than-one-billion-users-worldwide/) (Scroll down to "Reduced game stutter with high report rate mice") suggests that Microsoft has made some optimizations recently for high polling rate mice, but I'm not sure if they've been backported to Windows 10 or not.