LordOfMice/hidusbf #208: Hidusbf controller overclock

Thomas98767, 2023-03-22:
Does hidusbf need to be open at all times for your controller to be overclocked?

LordOfMice, 2023-03-23:
Driver can not be "Open" or "Closed". Driver can be "Loaded" and then "Participates in driver stack for device". And the technique that hidusbf uses (it is lower filter driver) requires the last.
On the other hand running Setup.exe is not required, of course. It is _setup_.

closed 2023-03-23 by LordOfMice

Thomas98767, 2023-03-23:
> Driver can not be "Open" or "Closed". Driver can be "Loaded" and then "Participates in driver stack for device". And the technique that hidusbf uses (it is lower filter driver) requires the last.
> On the other hand running Setup.exe is not required, of course. It is _setup_.

Why does it feel different when i have the setup application open or closed? Is it just a placebo?

LordOfMice, 2023-03-24:
It is.