LordOfMice/hidusbf #212: Unable to increase poll rate above 125 for keyboard

manuteswol, 2023-04-12:
I checked the documentation.... and I get an asterisk (*) next to polling rates above 125 for my keyboard.  These rates do not work.

Why is this?

manuteswol, 2023-04-12:
I saw the message.. and followed the directions.  Still see the asterisk next to the polling rate setting.  Wondering specifically what this means -- that it is a "low speed" device?

Found polling rate checks online to test keyboard.  The lowest two settings seem to take effect.  Settings at 125 and above do not.  Actually, the default 125 does not even seem to work.....

One strange thing is that I see "HID Keyboard" listed multiple times in the list of USB devices.

Anyway, thanks.  I will accept this keyboard cannot be OC'd or the associated system files are "too new", like stated in the documentation.

LordOfMice, 2023-04-12:
>that it is a "low speed" device?

Yes.

>and followed the directions

Once again - readme:
Recent info https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/ (the end of the first post and recent posts in thread)

manuteswol, 2023-04-12:
For someone that has been in tech for a long time.. I find the documentation cryptic and fragmented.

But thanks, I will try again.