LordOfMice/hidusbf #213: bInterval stuck on 4 for Scuf ReflexFPS

Marshyll, 2023-04-15:
bInterval will only goto 4 when overclocking my scuf reflexFPS.
[image]

LordOfMice, 2023-04-15:
1. I didn't see the full list of available rates nor result of [Copy IDs]
2. But I saw AMD USB Controller.

closed 2023-04-15 by LordOfMice

Marshyll, 2023-04-15:
Device:
"USB\VID_054C&PID_0CE6\9&128A4812&0&3"
"USB\VID_054C&PID_0CE6&REV_0100"
"USB\VID_054C&PID_0CE6"
  Name: USB Composite Device
  Driver: usbccgp
  Binary: \SystemRoot\System32\drivers\usbccgp.sys
    Path: C:\WINDOWS\System32\drivers\usbccgp.sys
    Version: 10.0.22621.1194 (WinBuild.160101.0800)
  Binary: HIDUSBF.SYS
    Path: C:\WINDOWS\System32\DRIVERS\HIDUSBF.SYS
    Version: 1.2.0.0
  LowerFilters: hidusbf
  BusSpeed: 2 (High)
  bInterval: 4
  Overrided.bInterval: 4

Controller:
"PCI\VEN_1022&DEV_149C&SUBSYS_88081043&REV_00\6&313998C&0&0140000A"
"PCI\VEN_1022&DEV_149C&SUBSYS_88081043&REV_00"
"PCI\VEN_1022&DEV_149C&SUBSYS_88081043"
"PCI\VEN_1022&DEV_149C&CC_0C0330"
"PCI\VEN_1022&DEV_149C&CC_0C03"
  Name: USB xHCI Compliant Host Controller - AMD USB 3.10 eXtensible Host Controller - 1.10 (Microsoft)
  Driver: USBXHCI
  Binary: \SystemRoot\System32\drivers\USBXHCI.SYS
    Path: C:\WINDOWS\System32\drivers\USBXHCI.SYS
    Version: 10.0.22621.1485 (WinBuild.160101.0800)

OS: NT x64 (amd64)