LordOfMice/hidusbf #214: struggling to overclock BT connected wireless controller

St1nGxD, 2023-04-16:
this worked for when it was wired, but i can't get it to work when its connected with BT... thanks in advance.

[image]
[image]

LordOfMice, 2023-04-17:
wireless part of communication channel can have (and usually has) its own restriction in addition to USB part

closed 2023-04-17 by LordOfMice

St1nGxD, 2023-04-17:
not following, what does this mean? what should i do?

LordOfMice, 2023-04-17:
Nothing. It's overclocking, sometimes it does NOT work. And usually wireless devices are not oveclockable because they are wireless.