LordOfMice/hidusbf #227: >this is an isolated incident

alexanderadreon, 2023-05-28:
>this is an isolated incident 

This is issue, but not with THIS software. The problem is not new and usually described by "after overclocking when I move my mouse I see higher CPU consuming". But most people don't understand well what they are actually measuring. And that is not overclocking of device (CPU consuming of overclocking itself is so tiny, that simply can not be measured at all by such tools as Task Manager) rather than reaction of some SOFTWARE to this overclocking (and if you move mouse over desktop this software is usually Explorer).

>but might be worth pinning this

You can not pin more than three issues and аnd now the hottest ones are pinned of course. But i will refer this on wiki page as typical example.

_Originally posted by @LordOfMice in https://github.com/LordOfMice/hidusbf/issues/201#issuecomment-1451463111_

alexanderadreon, 2023-05-28:
So what is the fix to this i keep reading but im not understanding how we fix the problem ?

LordOfMice, 2023-05-29:
I can not help with such errors because this is unrelated to THIS software. Try to insulate the problem yourself.

closed 2023-05-29 by LordOfMice