LordOfMice/hidusbf #234: Can’t delete hidusbf

Porknpig, 2023-06-17:
I’ve tried uninstalling hidusbf through the command promp with no success as hidusbf.sys still shows up in %windir%. In administered command promp I typed sc.exe delete hidusbf as shown in previous issues with a line that says installed service does not exist.. so why does hidusbf.sys show up in files? I’m still figuring out the issue on my controller being stuck on binterval 4 instead of 1. I’ve used your program for years with no issues until I reset my pc. I’m not very computer illiterate so I try my best to read through your pin posts and read me but i’m at a breaking point. Any help besides a pin to another post would be extremely appreciated. Thank you. 
p.s i really did try to solve this on my own it’s just not working.

Porknpig, 2023-06-17:
issue #132 did nothing to solve this problem. I added the upgraded system just in case for apex though. But still reading with filter on and rate set to 1000 Binterval reads 4 instead of 1.

Porknpig, 2023-06-17:
I also installed hidusbf sys with no luck aswell.

LordOfMice, 2023-06-19:
> I’m still figuring out the issue on my controller being stuck on binterval 4 instead of 1.

"filter on device" checkbox + MAXIMAL possible rate in rate pick (NOT 1000) -> [Restart] button .... If your rate did not change, then you should either reboot, plug-out and plug-in mouse cable...

>so why does hidusbf.sys show up in files?

Because without installed service it is just the file, not the driver.

closed 2023-06-19 by LordOfMice

Porknpig, 2023-06-19:
Thanks dad, so I was literally just tripping. I think i wasn’t running it correctly before or something. ps5 controller running at 8000 hz before was 1000 so I think that was the confusion. I kept trying to set it to 1000 instead of 8000. Sorry if I wasted any of your time but i’m sure it’ll help someone in the future.