LordOfMice/hidusbf #236: umap - Failed to communicate with the mapper / can you help me with this?

unkn0wnunkn0wn, 2023-07-02:
hi, i am using IntelliMouse Explorer 3.0 and i know it is overclockable 

so i tried to use your tool, it does downclock but not overclock

and i assume i have to apply hidusbfn because i am using Windows 11

so i tried to use umap but umap gave me error 

https://i.imgur.com/Zsg3Oi8.png

according to Overclock.net fourm, you have done overclock with umap

ofc i formmated USB drive with FAT32 and booted with it 

but cant get over this 

can you help me with this?

LordOfMice, 2023-07-04:
Probably you don't load umap .efi part.
It have to expressly print "Success" on UEFI boot screen on loading.
Note that you have to disable Secure Boot to load umap .efi part.
And the last - I never used Windows 11 myself with umap.

closed 2023-07-04 by LordOfMice

unkn0wnunkn0wn, 2023-07-04:
wait umap requires also secure boot disable? is that only once first time or everytime i boot computer, i should run everytime? 
so it still cant compatiable with game like valrorant or apex legends?

LordOfMice, 2023-07-05:
Yes, Secure Boot has to be disabled. There is some method (probably, i never used it) to load code with Secure Boot on, but it is not so simple as using unsigned code loaders like atsiv or umap.

unkn0wnunkn0wn, 2023-07-05:
LMAO even with secure boot disabled, it still doesnt load umap from uefi boot (i still cant see success on boot screen)

LordOfMice, 2023-07-06:
Check that you can load any other .efi application on your system. Something like UEFI Shell. And do not forget that for auto loading you have to rename .efi file and place it on right path on boot drive.