LordOfMice/hidusbf #244: frame drops when using ps5 controller on fortnite

Jordyk01, 2023-07-25:
Hi, recently I've been playing fortnite and overclocked my controller to 8000 pulling rate for my PlayStation 5 controller. However, when I play I'm able to keep stable at 240 frames as I should. Though every like 2 mins while playing the game I get sudden frame drops that go from 240 frames to 60 really fast then back up to 240. I've tried everything.... Im close to fixing this problem but just need that extra help. I've tried playing with no overclocking done to the controller and my pc feels normal ( no frame drops ) however when overclocked with the Hidusbf file I get random frame drops. 

Can someone please help me fix it, I feel like i'm so close to solving it.

LordOfMice, 2023-07-26:
Wiki:
Higher CPU consuming or some anomalies in software behavior after overclocking: https://github.com/LordOfMice/hidusbf/issues/201
You have to insulate the source of problem inside software you are using.

closed 2023-07-26 by LordOfMice

Jordyk01, 2023-08-02:
wdym by that?? im so lost