LordOfMice/hidusbf #250: Restart doesnt work. ERROR

eternalmeow1, 2023-09-08:
[image]
like i clicked on restart and this is what happend. Tbh im not sure if my controller is overclocked now or not. What should i do against this error

LordOfMice, 2023-09-09:
You have to read recommendation in message.

closed 2023-09-09 by LordOfMice

eternalmeow1, 2023-09-13:
wym?

LordOfMice, 2023-09-14:
The fact that you are not native English speaking DOES NOT free you from understanding messages. Use translator.

"You can try to disconnect and then reconnect device or reboot system."

eternalmeow1, 2023-10-15:
and if i reconnect my controller, will my controller be overclocked then?

LordOfMice, 2023-10-16:
Yes.

Fakeyy23, 2024-07-09:
I have the same issue. Rebooting pc or disconnecting device doesnt help and error remains. Only way I dont get the error is if i hit restart while having ds4 running.