LordOfMice/hidusbf #251: Mic not working when using hidusbf on w11, works on w10

Stxvie, 2023-09-16:
Overclocking my mouse and keyboard and even downclocking my mic worked perfectly on w10 (1607, 1803, 1809, 22h2), overclocking both keyboard and mouse works on w11 22h2, but ends up bricking my mic

From a bit of digging it bricks my mic as soon as the hidusbf service is running despite me not installing service on it, I even have my mouse on a separate usb controller and it still bricks my mic

Any help is appreciated, my bad if this was already covered in a closed issue

LordOfMice, 2023-09-18:
1. What is mic? USB microphone? Is it separate (from mouse and keyboard) device? But unlikely someone can downclocking microphone...

> as the hidusbf service is running despite me not installing service on it

2. Do not use patching versions of hidusbf (if you really don't need this).

Stxvie, 2023-09-19:
My mic is a budget USB mic, i've tested with a mic that connects through the aux port and it worked perfectly fine, every time I used hidusbf on w10 i've downclocked my mic every time with no issues whatsoever, 

And as stated I have my mouse on a separate usb controller with my other usb controller being connected to my keyboard and mic

Can you elaborate/explain what you mean by "don't use patching versions of hidusbf"? i don't quite understand what you mean by that

LordOfMice, 2023-09-19:
1. You have to answer my first question before anything else. Literally nothing to talk about without this.

>Can you elaborate/explain what you mean by "don't use patching versions of hidusbf"?

2. You have to read readme and study the content of \DRIVER\AMD64\ folder.

closed 2023-09-19 by LordOfMice