LordOfMice/hidusbf #268: XInputTest showing my ps5 controller at a max 300-400hz

mochaaaa11, 2023-12-22:
i overclocked my ps5 controller. it reads 0ms in ds4windows and shows 8000 interval with filter on in your program. however the controller feels really delayed and upon checking in XInputeTest, it confirms my feelings. the controller is averaging a 2.5ms on all tests.. this has never happened. ive had ps4 controller which arent as fast and they always showed 900-1000hz for polling and a avergae of 0.4-0.5ms in the tests... someone tell me how to fix please

LordOfMice, 2023-12-23:
>ps5 controller showing 300-400hz with 2.5ms

Why do you think that this device is capable for more?

>how do i fix this?

Ask the other owners of this device about they results in overclocking for the first.

closed 2023-12-23 by LordOfMice

mochaaaa11, 2023-12-23:
i use your program for a long time to overclock ps4 controllers. they read 0.5ms average with 1000hz polling rate when i test with input tester. the ps5 can go to 8000hz so basically 0ms - 0.5ms with a 8000hz... but it only reads about 500hz with 1.5ms to 2.5ms average. even in ds4windows when you overclock the ps5 controller, it shows 0ms. but the testing shows otherwise and i can feel the delay. feels like a stock ps4 controller without a overclock

LordOfMice, 2023-12-23:
>the ps5 can go to 8000hz 

You definitely make the mistake of mixing polling rate and rate the device is answering to the polls.

mochaaaa11, 2023-12-23:
> > the ps5 can go to 8000hz
> 
> You definitely make the mistake of mixing polling rate and rate the device is answering to the polls.

i enjoy the round of making me look like an idiot, but im not bringing this out of my ass..... when i overclock my dualsense it has higher ms in response time and lower hz than a ps4 controller. the dualsense should be getting lower than 0.5ms considering its being overclocked to 8000hz. no way that a ps4 controller is faster than the dualsense. makes no sense

LordOfMice, 2023-12-23:
>no way that a ps4 controller is faster than the dualsense after overclock. makes no sense

really? the speediest devices in the world (they CAN be overclocked to REAL 8000 Hz) - Microsoft mice on mlt-04 sensor are USB Low Speed devices so BY DEFAULT they run on 125Hz only. does that make sense in your logic?

mochaaaa11, 2023-12-24:
> really? the speediest devices in the world (they CAN be overclocked to REAL 8000 Hz) - Microsoft mice on mlt-04 sensor are USB Low Speed devices so BY DEFAULT they run on 125Hz only. does that make [...]

you may be miles more intelligent than me, i dont care right now because youre trying to correct on something without actually acknowledging the issue. i could still have a problem and youre more so just trying to make me look like an idiot. whether my terminology is right or wrong. numbers dont lie. the dualsense should be faster than a ps4 controller after overclock and its simply not and i want to understand why...... theres countless videos on YT "how to overclock ps4 controller", thats easy and makes sense.. you filter on device, put the 1000hz and install, boom you got a ps4 controller with 0.5ms and 1000hz...... BUT when you try the same with the Dualsense which allows for a high rate of 8000hz, it somehow performs worse with like 2.5ms average and ~500hz?.... i get you want to be smart, but maybe stooping down on my level can give us both something to understand here idk. it wouldnt be so weird if i didnt hear so many people say the dualsense is faster than the ps4 controller and the dualsense reads 0ms flat in ds4windows after overclock, meanwhile the ps4 controller shows 0.5ms in ds4windows after overclock. another weird reason i dont understand why the dualsense isnt overclocking correctly

mochaaaa11, 2023-12-24:
ill try to better ask my question cause you are pretty much the only one who could answer and i want to understand. its your program, you know much about this stuff. heck you call yourself the LordOfMice.. everyone says the dualsense can be pushed more, thats obvious, even you can agree. thats why you can go to 8000hz with it. Anyways, the older ps4 controller is great, you OC it to 1000hz and bang youre good with a 1000hz and 0.5ms average........ on the dualsense, youd think maxing the interval to 8000hz makes it even faster cause in ds4windows it says its getting lower latency, but in polling rate and latency tests, the dualsense is much slower. i just dont understand this with everything ive explained to you. i even tried to OC the dualsense to the same as the ps4, a simple 1000hz OC and till the ps4 controller was faster. that just doesnt make sense to me and i like to understand things.

mochaaaa11, 2023-12-24:
> really? the speediest devices in the world (they CAN be overclocked to REAL 8000 Hz) - Microsoft mice on mlt-04 sensor are USB Low Speed devices so BY DEFAULT they run on 125Hz only. does that make [...]

ill try to better ask my question cause you are pretty much the only one who could answer and i want to understand. its your program, you know much about this stuff. heck you call yourself the LordOfMice.. everyone says the dualsense can be pushed more, thats obvious, even you can agree. thats why you can go to 8000hz with it. Anyways, the older ps4 controller is great, you OC it to 1000hz and bang youre good with a 1000hz and 0.5ms average........ on the dualsense, youd think maxing the interval to 8000hz makes it even faster cause in ds4windows it says its getting lower latency, but in polling rate and latency tests, the dualsense is much slower. i just dont understand this with everything ive explained to you. i even tried to OC the dualsense to the same as the ps4, a simple 1000hz OC and till the ps4 controller was faster. that just doesnt make sense to me and i like to understand things.  

and i understand your logic.. but you kind of prove the point to my confusion.. why is a lower rated device (PS4 Controller) doing better than the Dualsense, if the Dualsense is higher rated and more capable.. the ps4 controller is doing better than the Dualsense.

put it this way using what you said... im seeing the Dualsense as the "speediest device", and i see the Ps4 controller as the "Microsoft mice on mlt-04 sensor"... how tf is mlt-04 sensor outperforming the speediest device in this case?