LordOfMice/hidusbf #270: polling rate downclock from 1000Hz to 240Hz, 120Hz

fuweichin, 2023-12-29:
**Why 240Hz, 120Hz polling rate? What's the problem with 125Hz, 1000Hz polling rate?**

Because we have 120Hz, 60Hz screens, and a polling rate which is not multiple of refresh rate may cause some jumpy reactive frames per second when mouse reports are coalesced and vsync-ed. According to online tool [Pointer Movement Timing Analyzer](https://codepen.io/fuweichin/pen/vYPEBNM) and post post [On mouse report rate (polling rate) and screen refresh rate](https://linustechtips.com/topic/1549729-on-mouse-report-rate-polling-rate-and-screen-refresh-rate/)

[image]
Using a 125Hz mouse with a 60Hz screen may cause 5 (which is 125 modulo 60) jumpy reactive frames per second

Also a high polling rate (e.g. 1000Hz) may sometimes overkill thus cause active frame dropping.

[image]
Using a 1000Hz mouse with Chrome causes active frame dropping.

**Any existing cases using 240Hz, 120Hz polling rate?**

+ Some phones/tablets come with 120Hz-polling-rate touchscreens and 60Hz screens
+ Apple Pencil uses a polling rate of 240Hz (on iPad Pro) or 120Hz (on Low Power Mode)
+ MabBook Pro (those with 120Hz screen) come with 120Hz-polling-rate touchpads.

**If minimum timer precision is 1ms, how to achieve uneven but acceptable 120Hz?**

Instead of keeping each polling interval 8.333333ms, repeat patterned polling intervals 8ms, 8ms, 8ms, 9ms, ....

For 120Hz report rate,
average interval: 1000/120 = 8.333333 = average(8, 8, 8, 9)

For 240Hz report rate,
average interval: 1000/240 = 4.166666 = average(4, 4, 4, 4, 4, 5)

LordOfMice, 2023-12-31:
It's impossible on modern hardware+software. But theory talks that you can use higher rates, like 1000+ Hz. And rate higher 2 times is enough.

>Using a 1000Hz mouse with Chrome causes active frame dropping.

It's the clearly problem inside Chrome, ask its authors about this problem.

>According to online tool 

And yes, use the RIGHT measurement tools.

closed 2023-12-31 by LordOfMice

fuweichin, 2023-12-31:
Apple seems to be aware of the problem since 2021, some MacBook Pro models (those with 120Hz-refresh-rate displays) successively come with 120Hz-report-rate touchpads.

For jitter-free media playback, VESA published MediaSync Display certification for displays, along with [AdaptiveSync Display](https://www.adaptivesync.org/).

A rational consmuer cares about diminishing marginal benefit, trys to find a sweet spot on quantity and quality curve. As a developer and a consumer I'd like to pay for vendors who help to find the sweet spots for us.

LordOfMice, 2023-12-31:
It is really good things. And people who play media on PCs well know about this tens of years, But how this realates to THIS software?

>A rational consmuer cares about diminishing marginal benefit, trys to find a sweet spot on quantity and quality curve.

Yes, it is true. But you have to calulate. 8,8,8,9 sequence has 9/8= 12,5% instability only. Of course, you can eliminate even such small things, but you definitely in the END of diminishing return, not in the begin,

JFYI, there is 8000Hz input devices...