LordOfMice/hidusbf #277: Amd Cpu Not working as good For overclocking as intel?

TheGr8t, 2024-01-10:


TheGr8t, 2024-01-10:
Hey so, I got a new laptop with an amd cpu and it seems like the overclock doesn't work well. There is Still delay.
What do i do?  Ik how it feels to have and not have input delay?
 But honestly, it's not a ''feeling''  it's more like I'm pressing an action at a precise time and it's not doing it as quickly as it should when the overclock is working.
 Im using ps5 controller, and the game im playing is rocket league at the higher ranks when feeling doesnt do anything if your reacting but your seeing your car not moving quick enough.

LordOfMice, 2024-01-11:
Yes, Intel controllers are better for overclocking, especially for more than 1K Hz of low and full speed devices. You can read more following link in readme.