LordOfMice/hidusbf #286: Cant uncheck filter box for overclocked controller

TrueTempo, 2024-02-22:
When I tried overclocking my scuf I put the bit rate to 8000 and filtered it on and installed the service however now my pc usb ports detect a powersurge and instantly shut off making it impossible to fix the controller no longer pops up on the app since the port being used protects itself. So now my controllers useless I've tried almost everything to get the program to completely reset but it never does I tested it with my second scuf. Always comes back up with the same overclocking settings do you have any advice on how to fix this?

LordOfMice, 2024-02-24:
It's somthing unusual. I know about this situation but do not know the source of problem still.

1. Try to set "with hidusbf" in "Devices" in Setup.exe
2. In any case read Uninstall section in Readme, you can always do manual uninstall.

ethayden97, 2024-07-13:
I was having this problem. I just used another usb port and my controller started working again

mahfoodh-nlip, 2024-11-12:
hello i need help eveytime i install fliter on i get driver error 
[image]

Owen2goodOHZ, 2024-11-21:
It says download not supported?