LordOfMice/hidusbf #294: Запутанное описание

ropucyka, 2024-04-24:
Инструкция, порядок действий и описание (комментарии, пояснения) намешаны в кашу и readme не лёгок в восприятии.
Что за sx64.exe также нет пояснения в readme.
Это замечание от заглянувшего в первый раз именно о подаче информации. Но да, скажет автор, первый раз всегда больно. :)

LordOfMice, 2024-04-25:
С каких это пор оверклокинг должен быть "лёгок в восприятии"? Он должен быть крайне неудобен или тут же появятся пострадавшие в большом количестве. И да, пример нарушения изложения порядка действий и "каши".

>Что за sx64.exe также нет пояснения в readme.

2. Это служебная программа (для работы установщика на x64 системах) и раз о ней ничего не сказано в описании - значит это вещь не пользовательского уровня.

closed 2024-04-25 by ropucyka

LordOfMice, 2024-04-26:
>И да, пример нарушения изложения порядка действий и "каши".

Третьего напоминания не будет...

> я пришёл немного насрать правды

Что ты пришёл насрать я заметил.  А вот про правду - пока не очень получается.