LordOfMice/hidusbf #298: Dual sense edge overclock

Zilla4k, 2024-06-12:
Hello wondering if you could help ive been trying to overclock my controller for some time now i de activated secureboot  turnt off microsoft driver blocklist and microsoft memory integrity all the steps but everytime i clickinstall service and click open another tab pops up and says installation failed any ideas what i could do to make this work, thanks.

LordOfMice, 2024-06-12:
1. There is no [Open] button in Setup.
2. In any case read readme about installing Service manually.

closed 2024-06-12 by LordOfMice

Zilla4k, 2024-06-13:
ive read read me and once you click install service it pops up with open...

LordOfMice, 2024-06-13:
>once you click install service it pops up with open

Check .INF file association

>ive read read me

No, you did not.  Because **installing Service manually** works **outside** Setup.