LordOfMice/hidusbf #300: Is there an option available to downclock a mouse to 125Hz on hidusbf?

Joshhhuaaa, 2024-06-15:
Hi, I'm interested in downclocking my mouse to 125Hz polling rate. Currently, hidusbf seems to only allow a minimum rate of 250Hz for downclocking, and I couldn't found any methods for downclocking to 125Hz.

If you're wondering why I want to do this, it's for compatibility reasons. Some older game engines only function correctly with a mouse set to 125Hz. In the past, I could adjust this setting using the mouse manufacturer's software. However, newer mice are phasing out support for 125Hz, leaving only options like 1000/2000/4000/8000 instead. Thanks.

LordOfMice, 2024-06-16:
> Currently, hidusbf seems to only allow a minimum rate of 250Hz for downclocking

1. For High Speed USB devices. The Low/Full Speed USB devices have 31Hz low limit. And there is such (rare) thing as USB Hub v1.x
2. It's possible to aggregate few HID reports in one with lower rate but this is outside of this software.

closed 2024-06-16 by LordOfMice

Joshhhuaaa, 2024-06-16:
Thanks for the quick reply. Could you point me in the right direction for aggregating HID reports at a lower rate? Since hidusbf can't do it, is there a particular software or method you recommend?

LordOfMice, 2024-06-17:
> Could you point me in the right direction for aggregating HID reports at a lower rate? Since hidusbf can't do it, is there a particular software or method you recommend?

Writing custom kernel mode driver or using some third party kernel mode input filtering software. Both methods require some programming skills. May be some direct solution exists by IDK it.
So using hardware solution is definitely simpler for ordinary user right now.

P.S. If you wish you can try to setup 125Hz (or lower) rate for High Speed device. 250Hz is the OFFICIAL lowest limit from MS and I did not check lower rates because I don't have High Speed USB Mouse still.

LordOfMice, 2026-09-30:
>And there is such (rare) thing as USB Hub v1.x

And not so rare ADuM3160/4160

LordOfMice, 2026-10-29:
The second (software) possibility is to use the driver of xHCI98 project.

Joshhhuaaa, 2026-10-29:
Luckily, I no longer have a need for this, but hopefully these suggestions will be useful to someone else in the future.

Unreal Engine 2 was particularly problematic with high polling rates, but since then, the affected games have been patched by me and others to support polling rates up to 8000Hz just fine.

LordOfMice, 2026-10-30:
>will be useful to someone else in the future

That's why i add this Issue to Wiki.