LordOfMice/hidusbf #301: GENERIC USB CONTROLER HOST NOT COMPATIBLE

hidusalx, 2024-06-22:
[image]

PLEASE HELP, i saw that issue#22 it was the same or similar issue but i dont rally understand what to do because im spanish and i dont too much english

hidusalx, 2024-06-22:
Before the red square was shown another error was shown first,it was this one
[image]

hidusalx, 2024-06-22:
i want to add that my secure boot is off,also de device i want to overcolck its a simple ps5 controller(the edge one)

LordOfMice, 2024-06-22:
In order to try:
1. [Install Service] button
2. Manual install of HIDUSBF.INF
3. Check .INF file association

closed 2024-06-22 by LordOfMice

KaiTheDestroyee, 2024-08-10:
How can I manually install HIDUSBF.INF / HIDUSBF_AS.INF
cuz my controller is completly not working after it is red 
[image]