LordOfMice/hidusbf #316: I need help, cant fix it by disabling hidusbf cause controller no connection

Gintas2002, 2024-08-02:
how do i fix it??? few months ago i played with this overlock on controller worked perfectly now today I tried connecting my controller to the same port I used the overclock and there's is no sign no connection??? how do I reinstal this overclock if it doesn't detect nothing???

LordOfMice, 2024-08-02:
>how do I reinstal this overclock if it doesn't detect nothing???

And how did you check this? And second, when you have some problems with THIS software you have device nonfucnctional but **visible**.

closed 2024-08-02 by LordOfMice

Gintas2002, 2024-08-02:
> > how do I reinstal this overclock if it doesn't detect nothing???
> 
> And how did you check this? And second, when you have some problems with THIS software you have device nonfucnctional but **visible**.

I just want to fix my controller... cause it doesn't work or detect ANYTHING when I connect it to the same port that I used this program with, like I said I did this overclock few months ago worked flawlessly, but haven't used it in a while and now when I connect it there's no detection so I cant uncheck what I did in the program cause it doesn't show there.

Gintas2002, 2024-08-02:
Hello?? how do i fix it??? how does this program even work, I removed that hidusbf.sys from windows32 its still same USB port no response with my controller, is there a regedit value I can delete to fix it or what do I do??

LordOfMice, 2024-08-04:
You DID not answer my question and contimue to talk with youself.
And yes, setup has option EXACTLY for fixing such cases and that is described in readme.

Gintas2002, 2024-08-08:
> You DID not answer my question and continue to talk with youself. And yes, setup has option EXACTLY for fixing such cases and that is described in readme.

I figured it out myself... Its was low battery and wasnt detecting controller but weirdly only on one port, the port i did this oc thing before...