LordOfMice/hidusbf #317: EA Anticheat doesn't allow overclocked controller now?!

LatvianGypsy, 2024-08-15:
Never had an issue with my controller until I updated Battlefield 2042 (8.0.1); now I can't launch the game because it says EA's anticheat is not allowing me to use my controller. Is this now considered cheating??

LordOfMice, 2024-08-15:
1. You didn't point your OS.
2. Refresh this software, the last update has significant changes in this area for recent OSes.
3. The last and main: You have to ask the makers of antivirus or game protection engines about they policies, not me. I'm supporting this software on **Microsoft Windows** and nothing else.

closed 2024-08-15 by LordOfMice

LatvianGypsy, 2024-08-15:
Everything is up to date but the issue persists. I did contact EA yesterday, so I hope they fix.
Thank you so much for this feature! ❤️