LordOfMice/hidusbf #323: Hidusbf safe?

Spikes6, 2024-09-19:
Is Hidusbf Programm safe for my Pc and how i make it with secure boot on because some game need to be on can i get the Support for that he know evrything abot that?

LordOfMice, 2024-09-19:
I assume you have to read readme.

closed 2024-09-19 by LordOfMice

Spikes6, 2024-09-19:
Just because it's in there doesn't mean it's immediately secure. Also, why isn't the certificate trustworthy?

Spikes6, 2024-09-19:
Are there any damages if you do it, disadvantages if you do it wrong and if you want to do secure boot with which regist key should you use the UpgradedSystem or WqhlSettings?

Spikes6, 2024-09-19:
and what is the difference between the two reg keys and what do they do?

LordOfMice, 2024-09-19:
>and what is the difference between the two reg keys and what do they do?
I assume you have to reread THE CURRENT readme. There is described the current restrictions and they are not the Scure Boot related.

Spikes6, 2024-09-20:
does it have a video tutorial about that?

LordOfMice, 2024-09-20:
The most recent manual is:
https://battlebeavercustoms.com/pages/overclocking
And you can get enough video manuals just googling "hidusbf". But pay attention that something might have changed since.