LordOfMice/hidusbf #327: mouse bInterval returning to default every restart/replug, polling rate also returns to default as per polling rate test

ReyesJayMilson, 2024-10-04:
1k polling rate works and all is fine except it resets every time I restart or unplug then replug the mouse. also default mouse has a bInterval of 10 but when it goes to 1 when its installed

LordOfMice, 2024-10-04:
bInterval to Defaultl only or Filter? to No too?

ReyesJayMilson, 2024-10-04:
[image]
Here is the default

[image]
Here is when its installed, filter on, 1000hz then clicked restart

[image]
Here is after uninstall service and unticked filter on but havent restarted yet

[image]
Here is after i click restart

LordOfMice, 2024-10-04:
>Here is after uninstall service and unticked filter on

And what are you waiting for after this? Overclocking still?

closed 2024-10-04 by LordOfMice