LordOfMice/hidusbf #329: still on 3.5ms delay after overclock

okekechuks, 2024-10-15:
Hi, i know this might be late but i have an issue where when i plug in my controller it is on 3.5ms delay, when i overclock it and check again it is still on 3.5ms. i had a ps4 controller it worked on, i got 2 more and they refuse to overclock
[image]
[image]
[image]
[image]

LordOfMice, 2024-10-16:
> i had a ps4 controller it worked on

On the SAME system? Then you just have unoverclockable devices.

closed 2024-10-16 by LordOfMice

okekechuks, 2024-10-16:
yes my pc has an amd ryzen 5 7535HS, will that be an issue?

LordOfMice, 2024-10-16:
You did not answer my question but asked your. Something went wrong.

okekechuks, 2024-10-16:
oh sorry about that, i first overclocked with my old pc (cpu was i5-8300H), the controller went from 3.5 to 0.5, got a new pc and new controllers, tried the already overclocked controller on my new pc and its still 0.5ms delay, stopped using the controller because of stick drift and weak buttons.

LordOfMice, 2024-10-17:
>tried the already overclocked controller on my new pc and its still 0.5ms delay

Ok, if some device is overclockable on some system and other device is not on the same system then problem with high probability is device, not the system. But! - do not forget about using USB hubs too.
In any case the better is to test overclocking of your new controllers on some other system (your old PC may be?), of course.

okekechuks, 2024-10-17:
what do you mean by using USB hubs?, also i just uninstalled all HIDUSBF.INF and reset the overclock status back to default. I want to restart the process.

okekechuks, 2024-10-17:
just did it again, i'll check if it worked
[image]

okekechuks, 2024-10-17:
did not work