LordOfMice/hidusbf #333: i need help

ExoKiru, 2024-11-15:
i been trying to fix this for 3-5 hours already i just want to play my game but, i recently reinstalled windows on my pc and tried to overclock it again, i tried everything on the youtube videos and nothing happened
[image]

MrDJ93, 2024-11-18:
ive got the same issue.

ExoKiru, 2024-11-18:
> ive got the same issue.

I can help you add me on discord exokiru i fixed it

LordOfMice, 2024-11-19:
It is HIDUSBF.SYS file not copied to %windir%\system32\drivers (probably caused by partially not working install script) or locked there by AV.

>on discord exokiru i fixed it

I assume it is better to share you expirience here, i can reference it.

closed 2024-11-19 by LordOfMice