LordOfMice/hidusbf #351: Can I now use hidusbf with Memory Integrity enabled?

ReelsFPS, 2025-06-04:
Hi!

I've been using your software for a long time now. It works great!
But, recently I stumbled upon an issue. I play both Call of Duty on PC with controller and Faceit on Counter Strike (with mouse n key).
I use your software when playing Cod only. 

Faceit Anti-Cheat requires windows to have Memory Integrity enabled. But, hidusbf requires to have it disabled.
This basically means I have to choose which one I want the most, I cant have both. 

Two weeks ago I noticed that there was an updated hidusbf file, thats says: NoPtach version is Memory Integrity compatible.
"Finally" I thought. Now I can use both Faceit and hidusbf. Well, the same issues happend. 

**My question is;**
Is this new hidusbf update made to work with Memory Integrity enabled? If not, do you know a way to have both Faceit AC and hidusbf?
Will there ever be a update that makes the software compatible with Memory Integrity enabled?

I've added a screenshot where it says the software is Memory Integrity compatible.

[image]

Hope to hear from you, LordOfMice.

Kind regards
Reels

LordOfMice, 2025-06-05:
> Is this new hidusbf update made to work with Memory Integrity enabled?

Yes. But only NoPatch version.

closed 2025-06-05 by LordOfMice

ReelsFPS, 2025-06-09:
I've been looking around, but don't understand how to get the NoPatch version. Do I have to download another software? Or is it just a setting in the original hidusbf? Would love to be using the NoPtach version.