LordOfMice/hidusbf #353: do i need these drivers if my mouse (a4tech w90 max) supports polling rates up to 2000?

tammee41, 2025-06-18:


tammee41, 2025-06-18:
need simple answer, i got 2 usb ports one of which is usb 2.0 the other usb 3.0 im using a laptop

[image]
(
generic usb hub
usb root hub
usb root hub usb 3.0
Universal Serial Bus (USB) Controller (xhci compatible) (AMD USB 3.0 Host Controller)
usb composite device
usb composite device
Standard Enhanced PCI to USB Host Controller
)

the mouse USB A connector is dark (is 2.0 its not blue)
when i switch update rates it moves smother and shows different results on polling rates in different apps ive tried so far (idk which idiot in that article suggest using mouse poll rate by oliver because it got several display bugs)

my main problem is it does seem like when i set 2000 or 1000 or 500 my pc or some programs in it start lagging as if they were skipping packets

even if i dont move mouse at all i feel like this is the lower update rate on usb port (drivers)

from specifications: one USB 3.1 Gen 1 (USB 3.0), one USB 2.0

heres the video of me switching between 125 500 1000 2000

[rec.zip](https://github.com/user-attachments/files/20788239/rec.zip)

[image]

tammee41, 2025-06-18:
you can close issue all i need is a yes/no answer and if yes which stage in the instruction should i follow and which driver should i install

LordOfMice, 2025-06-18:
1. IDK is your device overclockable.
2. I see AMD chipset on screenshot.
3. The link IS in readme: https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/

closed 2025-06-18 by LordOfMice