LordOfMice/hidusbf #359: Warnings during Controller Overclocking Process

MentalPow3R, 2025-08-26:
As soon as I click on 'filter on device' I get this warning (1).

If I click 'Yes' I get this other warning (2).

What’s the problem? How can I successfully complete the overclocking procedure?

I have Secure Boot disabled and Memory Integrity (HVCI) disabled.

[image]
[image]

LordOfMice, 2025-08-26:
You have to do exactly what the first warning recommends to do.

closed 2025-08-26 by LordOfMice

LordOfMice, 2025-08-26:
>I don’t know what this means, how exactly should I check what you said?

I assume it's obvious - after you will install service the warnings should disappear.

MentalPow3R, 2025-08-26:
@LordOfMice 
I never clicked "Yes" on the second warning because I thought it was unusual to get those warnings, so I didn’t continue.
Now I’ll do it.
If the rectangle turns red, it means the overclock didn’t work, and I’ll proceed with the 4 steps you mentioned in discussion #301.
If the rectangle stays green, it means the overclock was successful and everything is fine.
Correct?