LordOfMice/hidusbf #373: ds4 ms delay after overlock

zhrsarfraz, 2025-09-26:
[image]

as you can see i followed everything in the readme and after that i was able to put it up to 1000 but on ds4 my ms was still above 3 and its affecting my games can you please help me out thank you

LordOfMice, 2025-09-29:
Looks like you are literally ignoring ALL of my recommendations:

>Answer to the first one.

And answer to the SECOND.

>can rate LOWER

and show me the picture where you set rate to the HIGHEST possible value.

Why do you think that I wish to continue conversation of such stile?

zhrsarfraz, 2025-09-29:
my apologies but I can lower the rate yes and in that picture i set the rate to 1000 which was the highest value possible. i might know the problem tho. Usually i get my controllers from Sony but this time i ordered from DualShock and this one was kinda different and faulty so it might just be the controller

LordOfMice, 2025-10-01:
>but I can lower the rate

If you can set rate lower then it is just hardware restriction of your device (or USB controller).